adereth/dactyl-keyboard #1: Do you have a website / kickstarter etc for this?

jasonm23, 2015-08-06:
because it's awesome!

kastiglione, 2015-08-06:
:heavy_plus_sign: :one:

Nathan219, 2016-02-06:
:+1: Yes please

pupeno, 2016-11-16:
Agreed. We'll... That is if it was a full keyboard. I would buy it right away.

Eloren1, 2021-12-26:
Found them here https://ergohaven.xyz/keyboards

BTW, @adereth , does project license allow to sell them?

jaredjennings, 2021-12-27:
Adereth shows up occasionally but I haven't seen him here lately.

That site looks pretty nice. Not sure where in the world it is, but it seems buying across continents gets expensive, so although code flies around the world, commerce doesn't. The sites I know selling physical keyboards are ohkeycaps.com, diykeyboards.com, bastardkb.com.

See also https://reddit.com/r/ErgoMechKeyboards. There was a discussion there about licensing and sharing - can't find it now, will update later.

Forks of this code that I've heard of are the Dactyl Manuform (started by tshort, itself notably forked by carbonfet, and available as a web app at dactyl.siskam.link), the DMOTE (which begat the Concertina), the Lagrange, and of course my own fork, where I've tried many ideas, most of which have failed.

My experience 3D-printing this original Adereth Dactyl using PLA in a filament-squirting printer is that some parts of the model are thin enough that they break too easily when you put the keyswitches in. But I can't stop wanting to customize this two-piece design, rather than the Dactyl Manuform.