adereth/dactyl-keyboard #7: TRRS sucks

adereth, 2015-12-04:
I should pick something better.  Probably RJ45.

mathias, 2015-12-19:
RJ-11? https://www.sparkfun.com/products/132 +  http://www.amazon.com/5-7ft-Black-Telephone-Extension-Cable/dp/B0099L1H94/ maybe.

zorglups, 2017-08-24:
Yes. RJ11 would be a good ideabut probably more difficult for people to source/create.

On my Ergodox, I removed te sockets and soldered the cables straight to the board. Anyway... The 2 halves always travel in the same bag ;-)

I'll do a direct connection on the Dactyl too and this keyboard being less portable will even travel less.

My 2 cents.

TobiasWegener, 2018-04-14:
Hey everybody,
I am planning to build a Dactyl and was thinking about to use a HDMI cable (or VGA) connector and then directly combine the rows on both sides. These I will than directly connect to the Teensy board. 
Am I missing something or should that be fine?

gagbo, 2018-11-09:
Hi,

Why do you prefer running a video cable to a RJ45 ? Just curious at this point, I'm trying to plan a dactyl build

mr-ubik, 2019-02-24:
What about USB-Type C? I saw it thrown around on GH.

TobiasWegener, 2019-03-09:
> Hi,
> 
> Why do you prefer running a video cable to a RJ45 ? Just curious at this point, I'm trying to plan a dactyl build

Hi the reasons are that HMDI is an absolute standard cable (easy to get). Furthermore, it has 19 pins and thus makes a direct connection of the rows and columns with the teensy possible, which makes programming the teensy and the use of easy AVR and QMK as simple as for normal (not splitted) boards.

The problem is USB-Type C can be that it is unidirectional, so u have to be very careful that u don't mess the pin out when the plug is the other way around. I have seen people struggle with this.

And when size is a concern u can still use HDMI mini, which has all the benefits stated above.
I hope that answers your question. 
The biggest draw back with HDMI is probably that is still bigger than the TRRS.
If somebody has counter points I missed, please let me know. :)