adereth/dactyl-keyboard #8: Document PCB manufacturing and wiring

adereth, 2015-12-04:


jasonm23, 2015-12-10:
for reference https://github.com/adereth/dactyl-keyboard/pull/11

BubblesToTheLimit, 2015-12-10:
@adereth How much have you paid for printing the modell?
Printing this at some 3d printing service seems extremely costy (~ 600€)

adereth, 2015-12-10:
It comes out to $225 if you print it through Shapeways in their cheapest material: https://www.shapeways.com/shops/bespokeys

mathias, 2016-01-01:
I'm guessing the 220 Ohm (red red brown) resistors on the Teensy are these pull-up resistors on the Ergo Dox:

[image]

joedevivo, 2017-05-02:
The 220s look like they's supposed to be wired to LEDs, which aren't pictured anywhere. It'd be nice to call that out somewhere and talk a little about how you'd choose a resistor based on LED color. I tried, but just haven't been able to figure out how to turn Ohm's law into Red LED = 220 Ohms.

joedevivo, 2017-05-02:
I'm also trying to figure out the best way to wire it up to work as seamlessly with qmk as possible.

The good news is that at this point I'm just gathering parts, so I should be able to log the whole thing.