adereth/dactyl-keyboard #9: Add ALPS compatibility

adereth, 2015-12-04:


sbuller, 2017-09-08:
I see some ALPS code in dactyl.clj, but I can't really tell if this is currently enabled or not. Or how to enable if it is not. What is the status of this?

silverhook, 2017-12-28:
I see there is a Matias version of the LightCycle now, but not one for normal Dactyl. Could you please update that as well?

jaredjennings, 2018-01-07:
If you run the dactyl.clj that's currently in the repository, you'll get SCAD output with Matias-shaped holes in it. See the [single-plate](https://github.com/adereth/dactyl-keyboard/blob/e0416e329032b3648a86e53656906702603dbeb0/src/dactyl_keyboard/dactyl.clj#L48) vs. [old-single-plate](https://github.com/adereth/dactyl-keyboard/blob/e0416e329032b3648a86e53656906702603dbeb0/src/dactyl_keyboard/dactyl.clj#L21) functions (single-plate is what gets used later on in the code).

silverhook, 2018-01-08:
@jaredjennings Thanks :)

In the meantime I already ordered some Kailh BOX Jade switches to try them out in this board. Good to know i can fall back to my trusted Matias if need be.