adereth/dactyl-keyboard #12: F-keys

jahkeup, 2015-12-22:
Any chance a row could be added to the top for F1-12?

mathias, 2016-01-01:
There's a range here that controls how many rows the keyboard has: https://github.com/adereth/dactyl-keyboard/blob/master/src/dactyl_keyboard/dactyl.clj#L90

The trick would be that you'd need to add another row to the controller, the firmware, and the PCBs. The electronics seem very inspired by the Ergo Dox (but I haven't dug in yet) - so I'm not sure if the controller and the MCP23018 has enough "inputs" to add another row. Maybe someone who has tinkered more with the electronics can chime in here.

laialbert, 2016-01-31:
I tried doing just that and it does create another row for F keys, if you change the range to (range -1 5). However, I couldn't figure out how to change the back, right, and left walls to accommodate the additional row.

In looking into using the Teensy controller, it seems that you can probably accommodate a row of F keys (just barely). There's the Phantom Tenkeyless (https://deskthority.net/wiki/Phantom) keyboard that was built on the Teensy 2.0. If there aren't enough pins, then you'd need to use the Teensy++ 2.0, which has enough pins. (My understanding is that the minimum number of pins you need is number of rows and columns in your keyboard, plus 1).

Regardless, you'd have to adjust the keyboard matrix and update the firmware to match. I haven't looked into this enough to figure out how to actually do it.

linnemannr, 2016-03-23:
I did look into this with the ergodox electronics, and the dactyl is a variant of the ergodox with the same keycount. With the teensy 2.0 and the 16 bit IO expander, you do have enough pins to add an additional row and column to each hand and more. I too am looking in to adding the function keys. If I can grok clojure enough to parameterize the walls on the number of rows (and hopefully columns) I'll make a pull request. I can barely elisp right now so it might be a tall order :/ My 3d printer and gateron red keyswitches are beckoning...

plo009, 2016-11-11:
It would be great if anyone has made any progress on adding the function keys could make an update here, using a ergodox now and really think the function keys would be very handy.

PierreKircher, 2016-11-11:
why not just use a layer ? keybinding + numberkey = f-key usually the less keys the better

plo009, 2016-11-11:
Well for cases where you need to press more than just the function key, e.g. alt + f4 etc

PierreKircher, 2016-11-11:
even thats possible with mappings .. if you bind the switch of the num row to a different key .. say the "meta|win|apple" key on the thumb cluster  .. to archive that ..

pupeno, 2016-11-16:
I'd also like to have F keys, or even better, extra rows/columns of programmable keys.

nickhristov, 2016-11-22:
+1 please add another row.

mathias, 2016-11-22:
See https://github.com/adereth/dactyl-keyboard/issues/12#issuecomment-168314746 -- this is parameterized for the 3D model. You'll have to learn a bit about the Ergo Dox electronics and firmware to get that extra row in there and program it to be f-keys. 

I'd guess that pull requests for this are happily accepted and the community can help support it, so if you want it enough, I encourage you to give it a shot.

adereth, 2016-11-22:
Yes, pull requests are happily accepted.

Note that adding a row is going to require changing a bunch of other things and the naive approach of just adding the row will not work.  If you look at the current number row, you'll see that back of the caps are missing each other by about a millimeter.  With the current curvature, an additional row  will result in the caps overlapping, so that needs to be changed.

I'm planning on doing a build that actually has the number row removed and I'll use the function layer on the home row for numbers and the top row for F1-F12.  The work I'm planning to do to make it easy to remove a row will also involve me making the back wall's shape automatically adjust to the back row, so that should make this a little easier.

I'm not going to tell anybody how to live their life, but I highly recommend getting more comfortable with the function layer.  I'll point out that the thumb cluster on the Dactyl is much more accessible than on the Ergodox and with R1 caps on the top two keys, it's very comfortable to hold them down while pressing any of the keys on the main keywell.