adereth/dactyl-keyboard #16: Teensy firmware

iplus, 2016-02-10:
Can you commit code for Teensy controller according to your matrix?

arhi, 2016-04-21:
any news on this? what is the MCP chip used to connect the other side? MCP23017? MCP23S17? (i2c or spi or ?) are you using the same firmware as ergodox ( https://github.com/benblazak/ergodox-firmware ) or ?

BubblesToTheLimit, 2016-08-06:
I'm pretty sure he is using the ergodox-firmware (https://github.com/benblazak/ergodox-firmware ). He uses the same controllers (Teensy 2.0 and MCP23018 I/O expander) and the wiring that can be seen in his rough guide is almost identical (4/6) to how the matrix is defined in https://github.com/benblazak/ergodox-firmware/blob/master/src/keyboard/ergodox/matrix.h

EricGebhart, 2016-11-26:
@adereth  How different is it ? Why the modifications ?  The chips are indeed the same as ergodox. 

If it were ergodox compatible we could just use the QMK firmware directly.

What would it take to make it compatible with QMK ?

EricGebhart, 2016-11-26:
Presumably [this schematic](https://github.com/ErgoDox-EZ/docs/blob/557cbdaf66f672011447fcc757e8fecd92837648/ErgoDox%20EZ%20Schematic.pdf) could be followed and it would be compatible with ergodox-ez.

joedevivo, 2017-05-21:
I wrote up a whole thing about [Wiring a Dactyl for QMK/Ergodox-EZ](http://joedevivo.com/2017/05/20/building-a-qmk-dactyl.html) and I'm happy to roll some of that into the guide if that's helpful.

EricGebhart, 2017-05-22:
Nice article. Yes I think some documentation on wiring it as an ez would be great. The wiring doc is not very good as it is. And using the ez code makes sense.

Alevale, 2023-06-30:
> I wrote up a whole thing about [Wiring a Dactyl for QMK/Ergodox-EZ](http://joedevivo.com/2017/05/20/building-a-qmk-dactyl.html) and I'm happy to roll some of that into the guide if that's helpful.

Hey! Do you happen to have something at hand for a COMPLETE begginner? I would really love to build something to replace my 20-30 years old kinesis Advantage pro.. but I'm really scared of all of the wiring and not having enough info to build it myself with my limited knowledge (I feel like there's too many issues, hardware, software and hability to make it 😢 )

Many thanks!

Alevale, 2023-06-30:
> I wrote up a whole thing about [Wiring a Dactyl for QMK/Ergodox-EZ](http://joedevivo.com/2017/05/20/building-a-qmk-dactyl.html) and I'm happy to roll some of that into the guide if that's helpful.

@joedevivo Hey! Do you happen to have something at hand for a COMPLETE begginner? I would really love to build something to replace my 20-30 years old kinesis Advantage pro.. but I'm really scared of all of the wiring and not having enough info to build it myself with my limited knowledge (I feel like there's too many issues, hardware, software and hability to make it 😢 )

Many thanks!

jaredjennings, 2023-07-06:
@Alevale there has not been a great deal of activity on this Github repository or its issues in the past three years or so. You can find some electronics videos like [this one](https://www.youtube.com/watch?v=hjml-K-pV4E) on Youtube if you search "keyswitch matrix" or "handwired keyboard." People often use a pair of Pro Micros these days rather than a Teensy and an I/O extender chip. You can find people and ask questions at https://lemmy.world/c/ergomechkeyboards or geekhack.org. Depending on the physical condition of your Kinesis Advantage, you may be interested in https://michael.stapelberg.ch/posts/2020-07-09-kint-kinesis-keyboard-controller/