adereth/dactyl-keyboard #17: Getting one?

kastiglione, 2016-02-18:
I have a very uninformed question: are there any companies that would produce a dactyl keyboard from the sources and information here?

theelectricwiz, 2016-05-17:
I'm looking at the same thing. He's got the case on Shapeways http://www.shapeways.com/search?q=dactyl&type=

jonleung, 2016-07-21:
I would totally buy if that was an option...

ChunkLightTuna, 2016-07-21:
How much ya wanna pay?

kastiglione, 2016-07-21:
What's the market rate?

roy, 2016-07-22:
I think retail price should be around $300 to $400

jonleung, 2016-07-23:
That works for me : )

If anyone wants to build one, lemme know and I'd be down to buy it : )

StevenTammen, 2016-08-27:
Would also buy a keyboard if someone else would build it.

pupeno, 2016-11-16:
I'd also like to get one.

bthezebra, 2016-12-15:
I would also like one :)

roy, 2016-12-15:
I finally made one. Typing on it right now :)

bthezebra, 2016-12-16:
Hey Roy would you mind posting a couple of pictures of yours? I love to see your design. Did you end up going the hand wired route or did you cut a custom pcb?

roy, 2016-12-27:
Hi bthezebra, I went with the hand wired approach. Mostly because I have a little experience with soldering and none whatsoever with making a pcb, let alone a flexible one. Although it seems I used thick wires and the wrong trrs mounts cause the top and bottom don't fit anymore. I just bought a 3d printer so I can try out a couple things now for the bottom pieces.

[image]

[image]