adereth/dactyl-keyboard #19: Kinesis Circuit Boards

lewismoten, 2016-04-19:
I've got a couple Kinesis keyboards myself. I'm impressed at the similarities - although the keycaps are a bit smaller in height. Is it possible to put the circuit boards from the Kinesis keyboards into this one? I'm also noticing some missing keys - specifically the rubber function keys on the top row.

j9ac9k, 2016-04-22:
Considering the kinesis circuit boards are not flexible, I would imagine fitment here would be a tall order.  I doubt there is any way to say for sure without someone actually trying it.

laialbert, 2016-08-24:
The kinesis circuit boards for the keyless are indeed flexible. However, I would doubt that the circuit boards would fit as I'd think that everything would have to magically align.

PierreKircher, 2016-08-24:
they dont fit at all ... keywell spacing is different .. but you can create your own flex pcbs just like andereth proposed

adereth, 2016-11-30:
Sorry, but the Kinesis PCBs will not work in the Dactyl for several reasons:

1. The exact spacing of the switches is different.  I did not duplicate the Kinesis layout out of concern for legal liability.

2. The Kinesis PCB under the main keywells has no ICs on it.  All the logic boards are in a separate board that is housed in the center of the keyboard.

3. The Kinesis PCB extends beyond the keywells.  I wanted the Dactyl to have the smallest footprint possible.

closed 2016-11-30 by adereth