adereth/dactyl-keyboard #20: keycaps

arhi, 2016-04-22:
Hi,
any source for keycaps you can offer? I can get cherry blue keys I want from wasd keyboards for e.g. but they sell keycaps that are different for every row in the keyboard... if I look to get keys I have to chose the row info ... 

http://support.wasdkeyboards.com/customer/portal/articles/1366948-keycap-compatibility

[image]

so what to order?!

j9ac9k, 2016-04-22:
For keyboards with curved contours you generally want keycaps that are SA row 3, or DSA keycaps, I would imagine trying to use a cherry/OEM profile could be problematic, but then again, I'm speculating.

arhi, 2016-04-22:
thanks for the info @j9ac9k about R3 !! makes sense :)
as for DSA, never heard about those till now, quick google shows
http://pimpmykeyboard.com/dsa-pbt-abs-blank-keycap-sets/
that's something that would work good on dactyl ? any other source you could recommend? I get body from shapeways, I'll get keys from mouser/digikey/wasd, so it's keycaps that I need to figure out now where to get from :D and I'm having troubles locating a good source :(

j9ac9k, 2016-04-22:
Probably the most popular DSA profile is the granite keycap set: https://www.massdrop.com/buy/granite-keycap-set (orders are complete, you can usually find them second hand in /r/mechmarket ).

I can't imagine the DSA keycaps wouldn't work well with Dactyl, but only someone w/ one of these keyboards could say for sure either way.

arhi, 2016-04-22:
yeah would be cool to get info from someone that actually tried some profile on dactyl :D ... not sure why @adereth  didn't add to doc whet keys he is using (these on his pictures don't look like what DSA looks on pictures I'm googling now) ... looking at examples on http://keypuller.com/ what @adereth have on pictures looks like SA profile (looks like there's like whole bunch of different profiles for keys available, I had no idea!!)

j9ac9k, 2016-04-22:
Mechanical Keyboards are a very personal preference oriented hobby, there are loads of options out there, the right one, being the one you like.  

If you're new to the mechanical keyboard world, this project may be a bit of a big jump into it.  If you're still interested in knowing what typing on a contoured keyboard is like, consider purchasing a Kinesis Advantage keyboard (I have one, it's awesome!).  The number of options with the keyboard is very limited, however what you get in the mail is fully functional, so that may be a good place to start if you're not sure what you want just yet.

arhi, 2016-04-22:
electronics and 3d printing are "easy" for me :D ... and I'm making this one "for someone" ... I was also suggesting http://www.kinesis-ergo.com/shop/advantage-for-pc-mac/ for him but he insist in dactyl as I can make any custom layout for him and he needs that .. kinesis layout is not working for him (heavy emacs user) ... so I took on to myself to do it for him ... he's heavy mechanical keyboard user but is not interested in building, assembling etc ... while I'm builder :) who just use regular old split hand keyboards .. they work more then good enough for my need :) ... I'd personally first purchase few keys of different profile and test them myself but problem living in the middle of nowhere is that it's not that simple to get samples and shipping cost is killing a will to continue project :( 

thanks a bunch for the help!! reading now on the profiles looks like there's significant documentation available... I'll get some measurements so I can print keycaps myself and try how they feel before I order a full pack of keycaps :D (thankfully I make 3d printers so printing few caps would not be a problem :D )

vIiRuS, 2016-04-22:
Here is a parametric keycap that has configurations for SA (different rows) and DSA.
I'd personally go with the DSA, because I like the low profile they have and the symmetry of the key.

arhi, 2016-04-22:
found it few hours ago :D DSA (type 5) works perfectly

adereth, 2016-04-24:
I designed it to work with fully contoured SA key caps, but I've tried it with every profile and they all work well.  I may eventually do different designs that fit the lower profiles more tightly.

arhi, 2016-04-24:
@adereth thanks for the info, so the "glamour" image is with SA profile?

adereth, 2016-04-24:
Yes, that pic is all Row 3 SA caps, but I use fully contoured on my personal build.

ChunkLightTuna, 2016-05-28:
[image]
here's mine with DSA's from pimpmykeyboard

TartanLlama, 2017-03-10:
@adereth Do you have a list of the sizes and rows of the keys you use with contoured SA sets? I'm thinking about jumping on [this groupbuy](https://en.zfrontier.com/collections/keysets/products/sa-max-ashen), but only if it'd fill a Dactyl.

adereth, 2017-03-10:
I use R1 - R4 for the top 4 rows, then flipped R1 for the bottom.  I've used both R3 and R2/R4 for the 2u keys on the thumbs and everything from R1 - R3 for the 1u keys on the thumbs.  That group buy doesn't have enough 2u keys.  You may need to fill it out by ordering some blanks:

http://pimpmykeyboard.com/sa-2-space-pack-of-4/

zorglups, 2017-08-24:
@adereth I read somewhere else that people with larger hands should use dsa profiles (at least for the thumbs). Do you "confirm" that?
Hum... Could I ask... How long are hands... Say from the junction hand-wrist to the end of the longest finger.

Thanks.

TylerPham, 2021-05-06:
@adereth do you think mixing a sculpted SA for the alpha's and a set of DSA's for the thumb cluster would be more friendly for smaller hands/less stretching?

The sculp and the curve would reduce finger travel while the shortened DSA caps for the clusters would allow the user to keep a slightly more closed hand.

Also, do you think there is a significant difference between sculpted SA and uniform SA row 3 in terms of comfort/reach?