adereth/dactyl-keyboard #21: Which Unit Size .STL File mm, cm, inches?

thecmachine, 2016-04-23:
I'm trying to get 3d printer quotes in Manhattan and I have no idea what I'm doing as far as 3d printing goes.  Can Someone fill me in?  I'm going to try to compile a shareable google spreadsheet for 3d printing estimates on different Websites and share it on /r/mechanicalkeyboards.  

Please halp.

closed 2016-04-24 by thecmachine

adereth, 2016-04-24:
It's all in millimeters.  The spreadsheet is a great idea... it might also make sense to just include it in the documentation in the repo.

thecmachine, 2016-04-24:
I'll be happy to put the spreadsheet together.  I finally found an online tool that would tell me how large the plans were rendered at different scales and neither inches or cm would yield a Keyboard fit for anyone other than a giant.  Does anyone have any recommendations on the "least premium, most appropriate plastic/filament strength"?  I'm looking at higher infill/shell PLA as the lowest quality/minimum medium for printing, then including prices for ABS also.  Does anyone recommend establishing a higher quality material as a minimum strength print?