adereth/dactyl-keyboard #23: STL's in "things" and STL's on shapeways? same version

arhi, 2016-05-28:
I see that there were some updates on the "things" stl's but I don't see if shapeways already have the latest ones or only the original ones?
Talking about these from @adereth  https://www.shapeways.com/designer/adereth/creations

I can easily upload STL's from things folder from git, it comes out as same price, but I'd rather click to order ones uploaded by the designer :D

robertdale, 2016-11-04:
My bottom-left and bottom-right parts just got kicked back from Shapeways. I was following @adereth's advice for printing in PLA by uploading the templates myself.  I'm fairly certain I used the defaults in git. I'll re-upload to be sure.

Reason: Thin Walls
Description: This area will not print cleanly, if this is excess geometry, consider removing or cleaning up. For more information visit: https://www.shapeways.com/materials/pla
Find additional information in the attached image.

robertdale, 2016-11-04:
Just uploaded the originals and it complains about thin walls. But now they have a button for fixing thin walls. I'll let you know how that goes.

adereth, 2016-11-04:
I recommend using the "Print It Anyway" option.  That's what I did for my builds.  I would not use their automatic fixer.

Also, I found their PLA print to be extremely poor quality.  I recommend either using their "Strong and Flexible" material for a build that you actually want to use for a long time.  If you insist on PLA, you should try a local printshop instead.  Several people have had luck using 3Dhubs for that.

robertdale, 2016-11-04:
@adereth  They actually refused to build it after ordering and some preprocessing stage.  That's why I was trying the automatic fixer. But it doesn't seem to actually fix it.  Going forward, I plan on buying from your store for the final build. I just getting something cheap to experiment with.  I'd rather destroy the PLA prints before building the final version. In fact, I really only need the tops. The bottoms would have been nice to have to make sure I have the angles right but I hope I can compare to my Advantage.