adereth/dactyl-keyboard #28: FDM friendly model

Cka3, 2016-10-29:
Please consider making fdm friendly model, so more people can enjoy this beautifully designed keyboard. Thanks.

evamvid, 2016-11-04:
This is definitely fdm-doable (I'm printing the second half bottom as we speak), but there could definitely be improvements. 

The biggest thing would probably be doing the legs on the bottom half seperately. If there was some kind of captured nut mount on the bottom, it would reduce printing time, reduce the amount of support needed (saving time, quality, and plastic), and allow you to do the legs either out of a different material or with higher infill for more strength compared to the rest of the model.

danielfullmer, 2016-11-04:
In case anyone is interested, I made an alternative bottom piece which is much easier to print on FDM printers. See:
https://github.com/danielfullmer/dactyl-keyboard/blob/master/things/dactyl-bottom-right.stl

evamvid, 2016-11-06:
This is a much easier print. What could be done about the top section?

I printed one of yours (plus a right hand of the original), so I may end up with an asymmetric dactyl!

danielfullmer, 2016-11-06:
I should have mentioned, my fork is based on [kennykaye/dactyl-keyboard](https://github.com/kennykaye/dactyl-keyboard), which, among other things, uses 1.5 size keycaps on the outside column. I haven't checked if my bottom piece would actually fit the original adereth top piece.

roy, 2016-12-01:
@danielfullmer is it much work to do the same with the original? I'm currently making the keyboard as well and wiring by hand caused the casing not to fit. Your version would make it much easier.

danielfullmer, 2016-12-02:
The code is a bit messy at the moment, but I've tried to cherry pick my code onto the original adereth branch in [danielfullmer/adereth-new-bottom](https://github.com/danielfullmer/dactyl-keyboard/tree/adereth-new-bottom).

The stl files are these:
https://github.com/danielfullmer/dactyl-keyboard/blob/adereth-new-bottom/things/dactyl-bottom-left.stl
https://github.com/danielfullmer/dactyl-keyboard/blob/adereth-new-bottom/things/dactyl-bottom-right.stl

The kennykaye branch does something different with the clearances, so I can't give any guarantee that this will necessarily fit properly.  I haven't printed this version or anything.

I also use a slim hdmi cable to connect the two halves, so there is an hdmi cutout in the thumb cluster. It wouldn't be too difficult to change to a trrs connector if needed.

roy, 2016-12-04:
Ah that's very smart to use an hdmi cable. Although you miss compatibility with ergodox firmware then. Thanks for the other stl files, I'll check them out

tshort, 2017-03-11:
I've got a work-in-progress fork that works reasonably well on FDM printers. 

https://github.com/tshort/dactyl-keyboard

It works pretty well on FDM printers because the walls go straight up. Here are other changes in this fork:

* Thumb cluster from the [ManuForm](https://github.com/jeffgran/ManuForm)
* Parametric rows (4-6) and columns (5 and up)
* Parametric tilting and curvature
* RJ-9 for interconnection

To not clutter up this issue, you could post any follow-ups to the following issue at my fork:

* https://github.com/tshort/dactyl-keyboard/issues/2

I'm new to Clojure, but it's a lot of fun playing with @adereth's code and watch code build things.

joedevivo, 2017-07-14:
@danielfullmer how do you fasten the top and bottom pieces of your design together?

danielfullmer, 2017-07-14:
@joedevivo I currently don't have any fastening between the two pieces. It holds together fairly well for normal typing use. I had intended to figure out something more secure, but I'll probably end up doing something based on tshort's fork for my next version--which is just a single piece anyway.

joedevivo, 2017-07-14:
@danielfullmer ahh, the ol' gravity fastener ;) It'd be nice if there was a way for the pieces to clip together, since I like a good two tone effect. Mind posting some photos of how it looks assembled? particularly the inside. That base looks so roomy, so much possibility in there.

jrozner, 2018-06-05:
@evamvid could you and/or anyone else who has printed this with and FDM printer share some details about material used, printers, and settings? Working on printing this with PETG on my Prusa MK3 and about an hour in it's not looking terrible but it's also not looking amazing? Also, the amount of plastic being using to print the supports vs the model itself is a little concerning. Haven't done much with PETG yet and still dialing the the settings, especially for this manufacturer (eSUN).

jaredjennings, 2018-06-05:
@jrozner, I don't have my own printer, so I split it up in ~1.5 hour pieces. See jaredjennings/dactyl-keyboard. For most of the pieces, adding several layers of mandatory support, raft and brim, I've been able to rotate them up on a corner before slicing, with good results, using Hatchbox PLA.