adereth/dactyl-keyboard #29: Change the license for models from CC-BY-NC-SA to CC-BY-SA

silverhook, 2016-11-21:
CC-BY-NC-SA is **neither** a [Free Culture licence](http://freedomdefined.org/Licenses) **nor** an [Open Source Hardware license](http://www.oshwa.org/faq/#noncommercial).

The basic gist is that by restricting the _use_ to only non-commercial you prevent a substantial part of the community from _using_ it for whatever purpose – which in Free Software terms would mean it doesn’t give its users the most essential Freedom 0. In practice the license prevents (anyone except @adereth ) to sell Dactyl as a DIY kit or even just offer 3D prints of its case – whether this was intentional or not.

… or to quote OSHWA’s own FAQ:

>  _Why aren’t non-commercial restrictions compatible with open source hardware?_
>
> There are a few reasons.
>
> If you place a non-commercial restriction on your hardware design, other people don’t have the same freedom to use the design in the ways that you can. That means, for example, that if you and someone else both release designs with non-commercial licenses, neither of you can make and sell hardware that builds on both of your designs. Rather than contributing to a commons of hardware designs for everyone to build on, you’re limiting others to a very narrow range of possible uses for your design.
>
> In particular, because making hardware invariably involves money, it’s very difficult to make use of a hardware design without involving some commercial activity. For example, say a group of friends wanted to get together and order ten copies of a hardware design – something that’s often much cheaper than each person ordering their own copy. If one person places the order and the others pay him back for their share, they’d probably be violating a non-commercial restriction. Or say someone wants to charge people to take a workshop in which they make and keep a copy of your hardware design – that’s also commercial activity. In general, there are just very few ways for someone to use a hardware design without involving some sort of commercial activity.

I see the NC as a blocker for this otherwise awesome-looking keyboard to strive and would suggest to change the license CC-BY-SA-4.0, which is very similar and is still a copyleft license, but without the non-commercial restriction.

Erudition, 2017-04-01:
Agreed, please stick with free culture licenses!
I'm fascinated by the project and certainly considering building one... didn't notice the non-free license at first.

adereth, 2017-04-01:
I've removed the NonCommercial restriction and changed the license to CC-BY-SA.

closed 2017-04-01 by adereth