adereth/dactyl-keyboard #30: Schematic / differences from standard ergodox/ergodox -ez

EricGebhart, 2016-11-30:
Hello @adereth 

I'm about to put one of these together and I'm wondering if there is a schematic for the keys or how it differs from the standard wiring of the ergodox and ergodox-ez.

I'm considering just following the schematic for the ergodox-ez and using the ergodox firmaware on QMK.

So are their differences ?  Knowing them would make things a little easier.

Thanks,

Eric

adereth, 2016-11-30:
No, I wired it in the exact same configuration as the Ergodox, with the only exception being that the Dactyl doesn't have the inner columns containing 3 keys on each side.  I use Ergodox firmware on my builds.

EricGebhart, 2016-11-30:
Cool.  I saw the fork for the firmware so I wasn't sure if there were hardware differences driving the creation of the keyboard within that firmware.  

I've been creating my keymaps.  I use dvorak and bepo layouts so that is one of the reasons I want to use
the QMK firmware.

I'm starting to get all the parts, so I'll be assembling it soon.

Thanks for the quick response.

Eric

closed 2016-11-30 by EricGebhart