adereth/dactyl-keyboard #31: Running a group buy

ev0platypus, 2017-03-10:
I don't know another way to reach you so i am going this route.

I just built a this board and posted it up on reddit. I want to do a group buy and i wanted to see if you would have an issue with this. Can you contact me at EvoPlatypus@gmail.com? this is the only way i know how to reach you.

Also, i am adding in wires that are missing to the PCBs and making them into one big sheet. The goal is to supply them to the community so more of this keyboard can be made.

PierreKircher, 2017-03-10:
its opensource .. i doubt adereth has a problem with a GB thats as as if opensource demands it .. design credits are given ( same as with the ergodox ) .. you can reach him on irc .. he usualy hangs out in #geekhack on freenode

nickhristov, 2017-03-10:
+1 would buy

ev0platypus, 2017-03-10:
I just wanted to make sure as i haven't messed with this stuff before.
Better to be annoying than go in blind when it comes to money.

Haven't used IRC in years and even then i didn't use it that much. Was just
hoping to catch him on here but i guess it doesn't really matter.

ev0platypus, 2017-03-10:
I will be running a GB through GH after i get some prices. Keep an eye
there.

PierreKircher, 2017-03-10:
why not the regular massdrop way and let em handle it ? .. if there is plenty interrest .. ? ( after the former interrest check ofc)

the problem is buying from some unknown sources .. specialy the first independent group buy is kinder risky . - loads have been screwed over in the past ?

PierreKircher, 2017-03-10:
see issue 29 

https://github.com/adereth/dactyl-keyboard/issues/29

unless written consense you would actualy go against the licence

AustinBrister, 2017-03-10:
If it was mass drop I would be all over it.

ev0platypus, 2017-03-10:
I see. would you be willing to give me permission? or change the Licensing?

PierreKircher, 2017-03-10:
any andereth can do that .. and you need that in written form

ev0platypus, 2017-03-10:
I don't know how well MD would work with what i am doing though. I am
making the Flex PCB, or at least that is what i was planning.

PierreKircher, 2017-03-10:
they work with independend manufacturers .. but you just get payed once they delivered it .. - that way the customer remains safe .. and wont be screwed over .. on the other hand you can be sure there is a demand and you get the money as soon as you deliver your goods

ev0platypus, 2017-03-10:
Sorry i thought i missed something. I was replying to the email and i though it was him that was responding to that. My Bad.

ev0platypus, 2017-03-10:
Either way. I need to get ahold of him. So I will start working on that. 

This is my first groupbuy run so i am just going to blunder through it. Started trying to get the feel for who wanted it today. So before i go any farther i am going to get ahold of him. Hopefully.

PierreKircher, 2017-03-10:
so step 1 .. get the agreement 

- do your DD 

then interrest check on GH
MD runs there own poll .. once it makes it there

they presell em .. gather sales .. keep the cash as in an escrow ( pretty much that will be the case with all unknown entity unless you have valid a biz credit ranking ) ..

you get an purchase order for N amount of units - you take a credit against that purchase order .. - if you hold up your end of the bargain .. they relase the cash to you - all you do is ship N amounts to them .. rest is all done by them - safest way for any customers ..

PierreKircher, 2017-03-10:
tbh .. the manufacturing of the pcb's is the easyest part .. there are plenty pcb house that do flex pcb's .. there are more problems - hight ( arm rest required .. unless you have hovering hands anyway) .. the case is a pain to manufacture unless its shapeway ( VERY Expensive ) .. the only saving i can see is on the pcb's .. and in that case you can run a group buy just for them - my 2 cents

ev0platypus, 2017-03-10:
That is why i wanted to reach out to him was to make sure i wasn't going to
mess it up. The quotes for the Flex PCB i was getting for 1200. Dunno if i
was looking the wrong spot or what. I am sure bulk would bring it down but
i was just doing it for myself.

i know a guy by me that does wood rests. i am going to see if he wants to
join on this.

Again all of this rests on getting the agreement. I appreciate the help.
Kinda new in this space. Got into this hobby because i was bored at home.

PierreKircher, 2017-03-10:
http://dirtypcbs.com/store/pcbs

ev0platypus, 2017-03-10:
I saw that before, but it only goes to 10x10cm. I need upwards of 8x8
inches for the whole circuit. It is also 2 layers only which would force
you to insulate it. I might try spreading out the pieces to see what it
would be. A couple of the wires might be to wide.

PierreKircher, 2017-03-10:
tbh .. that board is 1 layer .. you wont even need 2 - and you have 1 pcb each side .. ya 10cmx10cm is small .. i agree but as i sayed there are plenty pcb houses in china that do them cheaply .. a mail to dangerousprototypes wont hurt ether ..- as i sayed i rather go 2 small flex + 2 hard pcbs for the thumbclustes .. just like kinesis did it .. seems to work for them over 10 years :)

ev0platypus, 2017-03-10:
That.... is a valid point.

I am trying to reach him through Shapeways. See if he responds there. If
not i will go at it.

Work for me tomorrow will be my DD to learn about licensing and the rules
of all of this.

Thanks again for your help! Time to be annoying to other people ;)

adereth, 2017-03-10:
Just sent you an email.  I feel like there are a number of changes that need to be made before doing anything large scale.

closed 2017-03-10 by adereth