adereth/dactyl-keyboard #40: Troubleshooting connection

postelrich, 2017-05-28:
Finished soldering my dactyl and flashed the ergodox-default qmk firmware using teensy loader. I tried connecting it to my mac and windows10 machines to no avail.  My mac did recognize that a keyboard was attached and brought up the keyboard setup assistant asking to press a button to the right of the left shift button.  Tried that and a bunch of other buttons but nothing happened.  Tried googling but didn't really find anything on how to troubleshoot.

postelrich, 2017-05-30:
Realized there was a link to the firmware here: https://github.com/BubblesToTheLimit/ergodox-firmware

Part of the right half works now, the left doesn't work at all. Progress!

BubblesToTheLimit, 2017-05-30:
Tips for debugging:
- solder on the LEDs which give debug information but are useless once you are done
- make sure you implemented the circuit diagram correctly
- talk to an electronics guy, there are lots of ways of debugging hardware which dont require dactyl-knowledge at all.
- measure the current while you test each switch to ensure your soldering was done right.
- if you are a beginner in building these things like me you are going to bang your head against the wall a whole bunch of times and you actually have to understand most things on your own. There are only so many tips people can give before they do the work for you which is what i experienced.

postelrich, 2017-05-31:
Thanks @BubblesToTheLimit I soldered on the leds and get a 1-2-3 light up when i plug the computer in and then they turn off. Is there a guide that says what the lights mean?

postelrich, 2017-05-31:
Also is there any debug logging that can be gotten?

postelrich, 2017-06-01:
The column to board wires were soldered to the wrong side of the diodes on the left hand and a couple on the right hand. Fixed that and its mostly working now. Some of the thumbpad keys still don't work.

PierreKircher, 2017-06-01:
its mostly again a resistor problem .. schottky diodes have a low voltage forward drop .. usualy around 0.15 and 0.4 V .. if you have a multimeter thats easy enough to test .. pay attention to the small stripe around the diods :)

postelrich, 2017-06-01:
Tried every single key. The left hand is working perfectly. The right thumb cluster isn't but I see some of wires on the buttons need to be switched. That and cleaning up the wires so the back fits and I should be done.

postelrich, 2017-06-02:
OK all buttons working now. Hopefully my logging will be helpful for someone else.

tldr:

- check row/column to board connections are on the right side of the diode
- the right hand thumb cluster wiring switches poles between the switches for some reason but it is correct, follow the pictures.

closed 2017-06-02 by postelrich