adereth/dactyl-keyboard #41: Clojure Source Code for the Light Cycle

pjones, 2017-05-29:
I would love to have access to the source changes that produced the Light Cycle SCAD files.

I want to experiment with a three (worst case four) row setup and also drop the thumb cluster to two columns.  I've spent a few hours playing with the Clojure source code and can't seem to get it to handle anything less than five rows.  It appears the code assumes five rows in several locations in subtle ways.

With the Light Cycle code I think I could use `diff` to figure out what was changed to allow for fewer than five rows.  At that point I'll submit a PR to make the design more parametric around the `rows` variable.

Thank you.

adereth, 2017-05-29:
Just checked in https://github.com/adereth/dactyl-keyboard/blob/master/src/dactyl_keyboard/lightcycle.clj

closed 2017-05-29 by adereth

SleepyBrett, 2017-09-12:
I'm confused about what the lightcycle is.

jaredjennings, 2017-09-21:
Ah - "It's a one-off modification of the Dactyl to support the limited set of keys that came in the LightCycle drop. It's also the first time I'm using Matias Quiet Clicks, which are really nice."
-- adereth, https://www.reddit.com/r/MechanicalKeyboards/comments/65ewt8/lightcycle_dactyl/dg9q1fd/