adereth/dactyl-keyboard #45: Guide: document the LEDs on PB5,6,7

ErinCall, 2017-06-21:
The circuit diagram shows 3 LEDs on PB5, PB6, and PB7. They don't show up in the example photos, though, and their purpose isn't clear to me. There's a note that "When the keyboard starts successfully two of the LEDs light up shortly." What does the third one do? Why do we need two LEDs to show successful startup?

Looking at [the ergodox-firmware source](https://github.com/benblazak/ergodox-firmware/blob/d40599230a38fbce1b670cbce561efc1ff0d8071/src/main.c#L57) (that function is defined [here](https://github.com/benblazak/ergodox-firmware/blob/d40599230a38fbce1b670cbce561efc1ff0d8071/src/keyboard/ergodox/controller/teensy-2-0--led.h#L34)), it seems they all turn on at startup. Is that the case?

Are they supposed to have any use beyond debug?

ErinCall, 2017-06-21:
Oh, and are the associated resistors necessary if the LEDs aren't present? They show up in the example photos, but I don't know if that's because they have to be there or because you left them in after removing the LEDs.

joedevivo, 2017-06-27:
@ErinCall, I'm halfway done with the qmk-guide in this repo, but you can see more about how I wired the LEDs [here](http://joedevivo.com/2017/05/20/building-a-qmk-dactyl.html) for now. With QMK all three will flash on startup, but you can make them do whatever you like in your keymapping. I have a function in my keymapping that lets me represent 0-7 in binary, so I can see which layer is currently active.

I don't know if the resistors are required without the LEDs, but I do know it's a pain to select them, as they're different for each color LED. I'm willing to try leaving the resistors while I test the new PCBs I'm working on [See #49], but I might not get the prototypes back for a while.