adereth/dactyl-keyboard #52: Printing lessons learned?

mathias, 2017-12-17:
Hi all,
Before I commit to about 24 hour print time on my home printer, can people share what worked and what didn't for slice settings on their machines? I'm guessing generate supports = yes,  especially for the top halves that curl away from the buildplate (these add about 10 h estimated to my print.) 

What about wall thickness? What is a reasonably-good thickness to ensure the key switches are sturdy and that typing won't break it. I've used 1.5-2.0mm thickness on other prints, but I'm wondering if I should do thinner here, or if that is a bad idea?

Obviously, a lot of settings will be different per printer.  I am hoping to use others' experience here a little and modify for my own printer. Thanks!

mathias, 2017-12-17:
I'm also curious whether you all use 100% infill or something less, and if the walls of the model are thick enough for that to matter. It's hard to tell whether infill is being computed or not on my old version of Cura that I use with my printer.

mathias, 2018-01-28:
I've now printed upper-right and upper-left files, and here's some thoughts. YMMV since printers are different. I was printing on my Monoprice Maker Ultimate, with PLA, and settings were (IIRC) 210 degrees F print, 60 degrees F heated bed.

The upper right file was first and I printed with 15% supports, 50% infill (which didn't seem to ever kick in) and a raft base (which seems necessary for smoothing out my prints.) The raft was a bit big on this one, and that took a lot of time. I printed at 0.15mm resolution, which I haven't used before, (previously I'd been printing at 0.1mm and the quality was great) in an effort to cut down on print time. Still, this took around 2 days of continuous printing to finish. I'd say somewhere between 24-38 hours.

The upper left printed at the same quality, but with much less support (10%) and much less raft. (I found the settings in my version of Cura to tweak the raft overlap and make it 0mm, since so little touches the buildplate -- or rather, the print is almost all overhangs but hte raft isn't needed where there won't be supports.) This printed fine, and will probably be easy to clean up -- the upper right piece is still waiting for me to finish pruning the supports off, and the 15% density of "line" supports are a bit too sturdy on this big print. With the speed, raft, and less-dense support settings, this print finished in about 20 hours.

Here's the two pieces on my desk, with only some of the upper right cleaned up:

[image]

caleb, 2018-04-30:
How did the keyboard turn out? I'm thinking about starting this project, and I've never 3D printed anything in my life, and I know that it would be cheaper and better to order from a professional, but it seems like a fun project to work towards.

mathias, 2018-04-30:
I have not printed the lower halves, as my 3D printer is broken for the time being. I do have switches, but have not worked on the circuitry (I was planning to wire point-to-point, as I've done on an Atreus keyboard build before the Atreus had a PCB.)

mathias, 2018-04-30:
It takes a very long time to print these, you may want to make sure you have a printer with a big enough build volume and that it is sufficiently fast. Some printers also can't run continuously for 20-30 hours for each build, like mine did -- they're just not that level of durability / reliability yet for some hobby-grade/DIY printers. YMMV, and good luck!

SteelBattalion, 2019-03-18:
I managed to print part of this out of pla with your recommendations. Some of the switch slots dead center didn't come out great but they should be use able. I'm planning on using polyurethane floor sealer to finish the case once its done printing. Just like you I had a component failure (thermistor) after the first print was done.

impaktor, 2019-03-19:
Would be interesting to hear thoughts on PLA _(what many use for their first print)_ vs. PETG vs. Nylon _(what adereth used)_. PLA has the downside of being brittle and very hard, which wouldn't absorb key presses the way one would like. Just considering the difference between aluminium vs. steel mounting plate for switches is quite noticeable, or so they say.

SteelBattalion, 2019-03-20:
So I managed to print the second piece and unfortunately it looks like it didn't print as well. There was poor layer adhesion around some points. I went ahead and gave both top pieces a layer of oil based polyurethane floor sealant anyway. I'll update tomorrow on how well it dried and if there was a noticeable improvement in rigidity.

SteelBattalion, 2019-03-20:
Polyurethane really helped the flex is nearly identical on both the good print and the one with layer separation after just one coat. I'm debating just sticking with one coat as I want to minimize the amount of hand filing and sanding I'll need to do in order to get the switches in place. Unfortunately my nozzle clogged some time this am in the middle of printing one of the bottom plates.

SteelBattalion, 2019-03-21:
> Would be interesting to hear thoughts on PLA _(what many use for their first print)_ vs. PETG vs. Nylon _(what adereth used)_. PLA has the downside of being brittle and very hard, which wouldn't [...]

It's not that straightforward to compare against nylon. Since there are so many blends. I feel like we'd have to all agree on a set of commonly available types and each take responsibility for making a set...then meet at a convention to have a fair comparison.

SteelBattalion, 2019-04-04:
I finished the print but thought  i'd finish the build prior to adding more comments. First it might be worth adding a bit more suppoprt structures and deal with a more painful and tedious removal of the supports. Second for the less experienced at soldeing and 3d printing perhaps printing out an extended bottom case would be the way to go. The extended bottom case found here. https://www.thingiverse.com/thing:2436848  is what I ended up using and I quite like it.

SteelBattalion, 2020-10-18:
It's been over a year since I've printed my PLA keyboard and coated it in polyurethane. The keyboard has lived on my desk as a daily driver since then. I worked from home even before the lock down and it's held up well to typing. I don't use it for gaming so it doesn't get the type of button mashing abuse that some games require. Strictly as a tool for typing it's great. I also added six 3d printed buttons with micro switches to the base of my keyboard via double sided tape. Shortly after the initial print. Those buttons did not get a coat of polyurethane however as they get very little use they are also holding up just fine. I haven't dropped the halves nor had anything drop on them so I can't comment on drop test durability, but for typing my print is holding up very well. 

[image]

Here is the model for the additional buttons I ended up using. 

https://www.thingiverse.com/thing:1909644

SteelBattalion, 2020-11-01:
For those that might be curious my original print required some modification to both the print and the amoeba switches. Nothing a dremel tool and a soldering iron can't hash out. 

[image]
[image]