adereth/dactyl-keyboard #53: File not rendering in openSCAD

sevinty7, 2017-12-27:
Tested the same file from the things folder in openSCAD for windows and openSCAD for ubuntu. 

Sits at  "Rendering Polygon Mesh using CGAL..." with 0/1000.

Please advise.

jaredjennings, 2017-12-27:
I saw that same sort of behavior, and chanced to leave it alone once. After like five minutes, it jumped to 400 and started counting. Have you tried leaving it to run for what seems to be an obscene amount of time?

If you don't want to run the Clojure code you can look at the rendered STL files; if you do run the code, you can turn the detail down to get an scad file that will render much faster.

Hope one of these helps; let me know.

sevinty7, 2017-12-27:
I'm planning on modifying the dactyl into a mini dactyl, so I'm using the clojure code to do that. I'm not sure if my laptop cpu just can't handle it, but i left it for 30 minutes with no success.

sevinty7, 2017-12-27:
I found the resolution setting and lowered it to 1. The inital render loads much more quickly, but when I try to press f6 so that I can export it as an STL file it does the same thing as before. Let me know if you guys have any ideas.

sevinty7, 2017-12-27:
Also this is what the inital render looks like when I download the openSCAD file from github. 

https://i.imgur.com/sENxsUa.png

sevinty7, 2017-12-27:
@xcode-manolo I'm wanting to drop the amount of keys down to 32, only 2 thumb keys, and have it printable in a very small area 120mm^3.

xcode-manolo, 2017-12-27:
Ah, that's pretty small.  I had think only in removing numbers row.  That's
enough for me !

Thanks Derek !

jaredjennings, 2017-12-27:
i also don't have a large enough print volume for the whole thing, but i've set out to print a full-size keyboard in pieces and glue them together.

don't miss the lightcycle.clj; it might be more customizable for small keyboards.

as for the openscad screenshot, you have me baffled.

sevinty7, 2017-12-28:
I'm wondering if the render is just failing on my laptop core i5. I'll try it on something with more power at some point.

sevinty7, 2017-12-28:
Tried it on another computer. The preview showed up correctly, but the file did not render. I had left it for about 10 minutes.

orcinus, 2017-12-28:
Have the same issue. Sits at 127/1000 for ages (on a 3.6 GHz i7 core).
Tried leaving it for 20 minutes or so, hasn't moved.

Additionally, the pre-generated STLs are full of issues that cause various slicers to do weird things, even after running it through the Netfabb cloud service.

orcinus, 2017-12-28:
Found a solution. Well, kinda.
It renders fine in old OpenSCAD (2013-06 for example), except it takes about 2 hours or more.

Also doesn't have problems with self intersections and malformed polys either.

orcinus, 2017-12-29:
Scratch that last bit.
Some parts still have self intersections.

mcdviii, 2018-08-08:
So these are issues with OpenSCAD updates, specifically?

orcinus, 2018-08-08:
Yes and no.
The parametric geometry isn't too clean, and OpenSCAD is notoriously bad at doing booleans on more complex geometry, often crashing and resulting in self-intersections.

Since the algorithms changed from version to version, some have more luck with some edge cases, others with some others.

mcdviii, 2018-08-09:
Is there something more robust & stable that the project should be ported to?

jaredjennings, 2018-08-09:
Judging from the searching I've done every time I hit a roadblock, OpenSCAD and its underlying CGAL library are the only thing in this area with staying power. libfive looks interesting but has no hull function, upon which this code depends a lot.

orcinus, 2018-08-09:
Antimony is interesting, but i think it'd be a bit more of a pain to work in, as you'd need to write a ton of missing semi-primitive stuff first. Fun fact - the author of Antimony works at Formlabs.