adereth/dactyl-keyboard #54: diode direction - Row / Columns Drive

funkrider, 2018-01-17:
In the README.org file the hand wiring pics as well as the text state "As you can see in these following pictures, I went for the “row-driven” setup (visible by the direction of the diodes, the black line being towards the key-switches). ". However it is my understanding that the black line as the cathode pointing towards the key would indicate the "Drive Columns" type setup would it not? 
I think that the firmware has a simple config setting for either selection however that would need you to re-compile it for "Drive Columns" option. Am I correct or am I just a n00b out of his depth?

raychengy, 2018-04-01:
I believe you are right. If the black line on the diode is pointing towards the switch, its "Column" driven. The columns will be high and the rows that will be read as inputs. 

If you're using the [benblazak/ergodox](https://github.com/benblazak/ergodox-firmware), you can switch the row/column driven option from the `/src/keyboard/ergodox/option.h` file.

raychengy, 2018-04-03:
Sorry, I'm wrong in my previous statement. I've wired the diodes such that the cathode is pointing towards the switch which is on an ergodox pcb board the circular solder pad, therefore its `DRIVE_ROWS`.