adereth/dactyl-keyboard #55: 3x5 key layout

kevinastock, 2018-02-03:
I'd like to print a version with the main part of the keyboard only having three rows and five columns per hand ('q' to 'b' on the left, 'y' to '/' on the right). However, modifying the rows and columns at https://github.com/adereth/dactyl-keyboard/blob/master/src/dactyl_keyboard/dactyl.clj#L117 is insufficient - it seems some parts are hard coded to their size independent of those variables. Could you please provide some pointers as to which parts of the code need modified to produce a correct 3x5 keyboard? Thanks!

matt1003, 2018-02-12:
tshort has done something similar with his dactyl-manuform fork of the dactyl-keyboard. Perhaps he may be able to provide some advice. See https://github.com/tshort/dactyl-keyboard

xcode-manolo, 2018-06-11:
Hi @kevinastock !

What a great idea !  A MiniDactyl !
Are the STLs and the QMK firmware on Github, do you know it ?   
I hope so !  
It would be fantastic !

Have a great day !!!
;-)

fi0, 2018-06-29:
Do you want one row fewer than light cycle?

kevinastock, 2018-06-29:
I'm not familiar with light cycle. As mentioned above, it would be similar to https://github.com/tshort/dactyl-keyboard, although without bottom row of three keys, and I would also be modifying the thumb cluster to be similar to keyboard.io

manolodeinternet, 2019-06-25:
Hi @kevinastock !

Mini Dactyl already exists !!!

[Click here for Reddit post](https://bit.ly/2wS1JOM)

I think that you are trying to get a keyboard similar to this one.  Or a variant.
Maybe u/crystalhand on Reddit could help you achieve your goal.

Good luck !
;-)