adereth/dactyl-keyboard #56: Kailh is so hot right now, switches don't fit

joedevivo, 2018-03-02:
That's a Mod-H on the left and a Kailh Burnt Orange on the right.

There's a notch in the base of the standard MX switch that keeps the switch snug in the Dactyl case, but the Kailh switches are basically square.

[image]

FSund, 2018-03-14:
Can anyone confirm this? I was considering a Dactyl, but was planning on using Kailh BOX switches. 

Do they not fit in the hole at all, or are they just not properly secured? One could always hot glue them in if they aren't properly held in place.

jaredjennings, 2018-03-14:
Judging from the picture, and from what I've seen of the switch hole design, the Kailh switches would likely not fit. This is quite important to me, because I've chosen and bought Kailh Brown switches for my build. Most simply, the pieces on the sides of the Dactyl switch hole that engage these Cherry-specific notches could be removed, and then glue may be required. I'd like to find a way that doesn't require glue. But in my modifications I haven't gotten to this problem yet. If no one has solved it by the time I arrive at it, I will, but I can't promise exactly when that will be.

chav02, 2018-03-15:
I've got a newly printed dactyl-manuform and kailh switches right here, and can confirm that their new design will not fit. :(

mcmadhatter, 2018-03-17:
I was able to get burnt orange box switches to fit in my dactyl-manuform, but they only fit in if the the little notches are set to deform the area around the led hole, I would still want to glue them in as well, just for extra hold.  Kailh Pro's also fitted.  (ignore the horrible yellow primer, I am in the priming/sanding phase)

[image]
[image]

FSund, 2018-03-27:
To make the Dactyl compatible with Kailh switches you can just edit [line 37 of src/dactyl_keyboard/dactyl.clj](https://github.com/adereth/dactyl-keyboard/blob/master/src/dactyl_keyboard/dactyl.clj#L37) and remove the `side-nub`. Replace

    plate-half (union top-wall left-wall (with-fn 100 side-nub))]

with

    plate-half (union top-wall left-wall)]

and you should be good to go.

I've made a fork of the [Dactyl-ManuForm](https://github.com/tshort/dactyl-keyboard) where I've implemented that change, as well as removed the teensy holder and wire posts, as well as some improvements suggested by [jmg123 over at Geekhack](https://geekhack.org/index.php?topic=88576.msg2578041#msg2578041). The diff can be seen [here](https://github.com/tshort/dactyl-keyboard/compare/master...FSund:master). This model should fit Kailh switches, and print faster.

TobiasWegener, 2018-04-14:
Ah, was planning to use Kailh Box Jade switches, or to they already have them...
Hm, I think I will go the manual way and remove the notch by hand may be with a sharp knife.

fi0, 2018-04-20:
It seems that oetumu cannot be used with side-nub. But gateron can be fitted.

Sylk, 2018-11-08:
@adereth please close this.