adereth/dactyl-keyboard #57: Incorrect pinout with the Teensy & MCP diagram in guide/circuit-diagram.png

raychengy, 2018-04-01:
Based on the ErgoDox EZ [schematics](https://github.com/ErgoDox-EZ/docs/blob/master/ErgoDox%20EZ%20Schematic.pdf), pin `INTA` on the MCP23018 is not used. I think there is a simple mistake on the [guide/circuit-diagram.png](https://github.com/adereth/dactyl-keyboard/blob/master/guide/circuit-diagram.png) file. Could you tell me what program was used to make this diagram? Col-5 to Col-0 needs to be shifted up a input on the MCP23018 schematic.

[image]

Mese96, 2018-04-01:
These pull request would fix this:
https://github.com/adereth/dactyl-keyboard/pull/51

raychengy, 2018-04-03:
Awesome! I'll close out this issue.

closed 2018-04-03 by raychengy

nomath4u, 2019-02-26:
This never got pulled in.

Marcus-Zhu, 2019-07-26:
Please consider pulling this fix.

nomath4u, 2019-10-26:
Just started making my second one and made the same mistake again 😭