adereth/dactyl-keyboard #58: Pull parameters out of dactyl.clj into their own params file

rmwphd, 2018-04-11:
The parameters of interest for customizing the shape of the keyboard to suit a given user are not well-exposed to the user editing the code. It would be nice to create a user parameters file which contains all of the parameter vars and then import that file into the dactyl.clj file. 

I guess from a basic search that, in Clojure, the correct thing to do would be to `use` or `require` the a new namespace called, e.g., params from a params.clj file in src/params.

I'd do it myself, but I can't easily tell which parameters ought to be factored out.  If someone can explain to me what variables decide, e.g.,  the angle of tenting or the curvature of each finger column, I'll be glad to write this up and put in a pull request.

jaredjennings, 2018-04-12:
The tenting angle is hard-coded: it is the `(/ π 12)` toward the end of `key-place`. 

The curvatures across the keyboard and from top to bottom are defined as α and β. I don't know how to express what these do: I just changed them and re-rendered a lot of times, and when they are bigger the keyboard is curved more tightly. This iterative exploration goes a lot faster if you comment out the case in the `write-scad` calls at the bottom.

nickcoutsos, 2018-05-03:
Yeah, alpha and beta define the angle between rows and columns. What really makes customization awkward is that the radii of the curves are defined in terms of those angles as well as keycap dimensions. Documentation would be fantastic for people making variations that are still "correct" implementations of the ergonomics but I don't know who is interested in maintaining anything like that.