adereth/dactyl-keyboard #60: Bad Model?

jrozner, 2018-06-07:
I'm working on printing the dactyl case pieces at home on my Prusa Mk3 using Slic3r Prusa edition to slice the file. I'm seeing some weird behavior in the left bottom with teensy when I slice and print it. It looks like there might be some bad geometry that's causing it to be seen as two pieces rather than one and basically builds supports between the two which snap apart when I rip off the supports. Is this intended and if not is there an issue with the model? In the screenshot below the yellow is the model being printed and the green are supports. You can see in the top right corner there is a section of supports that cuts through it.

[image]

When I import the stl file into slic3er it reports that there were 12 errors that were auto repaired but I'm not sure how to see what those are. Has anyone else experienced this?

[image]

mcdviii, 2018-08-08:
I am experiencing something similar in roughly the same area on the nonstandard models under /things. My problems are seemingly missing mesh. I even built from OpenSCAD source. 

I'm away from computer right now. I can post version numbers later.

jrozner, 2018-08-09:
I actually tried a few prints flipping it 180 so that the outer shell would have a better finish. I had issues getting a successful print with that orientation due to the minimal surface area touching the build platform. The corners came loose and destroyed the print.

RedWagon, 2018-11-07:
I'm having issues with the left side with teensy as well.  I compiled my own STL and repaired it in slic3r and got something that looks manifold in the preview but I'm getting the issue where supports just cut through the model. 

Printing discussion:
This was printed upside down.  To fix the corners I set enforce supports for the first 10 layers. 
https://www.thingiverse.com/make:561129

jaredjennings, 2020-10-19:
https://www.reddit.com/r/openscad/comments/gmpnjj/question_is_anyone_else_experiencing_issues_when/