adereth/dactyl-keyboard #61: Thumb cluster alternatives

fi0, 2018-06-10:
Has anyone seen a dactyl fork with a single row 1u keys? I plan to use two to three keys per thumb.

CreamyCookie, 2020-06-21:
@fi0 Maybe not what you're looking for, but @ibnuda created a version with an all 1u thumb cluster, which looks great and makes sense to me, since I tend to only hit the 2u keys at the very top, so why not get two additional keys for free. 

His site is http://dactyl.siskam.link/ where you can configure your own Dactyl / Dactyl Manuform. 

Actually, I just checked and you can have just two keys per thumb: 

[image]

fi0, 2020-06-22:
thank you @CreamyCookie

closed 2020-06-22 by fi0