adereth/dactyl-keyboard #64: Mini Dactyl

xcode-manolo, 2018-06-15:
Hi, guys !!!

Does anyone know about a Dactyl with 3x5 +5 keys ?

It would be a kinda of a MiniDactyl.  A minimal version.

Have a great day, you dudes !!!

;-)

szos, 2018-08-24:
the dactyl manuform has a 3x5 +2 under row 2 and 3. the thumbs may turn you off though. you can also grab the clojure environment and define your own.

xcode-manolo, 2018-08-24:
Yes, I know that model.  But those two extra keys... I can’t find sense for them.
And I know, the clojure sources are open source, but the problem is the omnipresent problem: time.
It doesn’t worth to me learning clojure for modify sources.
I was looking for a model as I dream, I know, it’s very difficult that exists, but it isn’t impossible.  That’s because I asked the community about the existence of that model.

But at the end, I have to finish typing on a Mini Dactyl.  If it doesn’ exist, I will have to give in and create it, I’m sure.

Thanks anyway, dude !!!

;-)

manolodeinternet, 2019-06-25:
Hi guys and @szos  !

Mini Dactyl exists !!!

[Click here for Reddit post](https://bit.ly/2wS1JOM)

I finally got it !

;-)