adereth/dactyl-keyboard #65: Keyboard with return row

jgwinner, 2018-10-25:
I've been using a keyboard for MANY years.

I love the parameterized design files; I'm wondering if I could use these to build a keyboard with a return row on the right hand side.

I realize there's better ways to do it, but I use a laptop keyboard fairly often, on the go, so I can't drag the keyboard around everywhere. I'm just sort of used to it. 

Is that possible or would it make the two sides too big? 

I have to admit I can just try it, but thought I would float this out here first.

I know I should use my thumb :) but I know that pinky will occasionally reach out - especially for the backslash. 

Also - it would be great if on the front page there was a diagram showing what key combinations there are; as a n00b to these kinds of keyboards it's a little confusing.

Thank you for listening,

== John ==

jaredjennings, 2018-10-26:
John, you might want to see pull request #4, which someone was talking about the other day. It has an extra column.

As for choices - the Dactyl is defined using code. I tried to make my fork have code and separate configuration, so the choices would be obvious, easy to make, and easy to "deploy," i.e. make pictures and STLs for, or make [lists of](https://www.xkcd.com/974/). I haven't even accomplished that goal for the choices I've added in my fork, much less the 90 other forks, much less whatever branches or pull requests they may have.

I started once to try to make a time-lapse video of the changes I've made in my own fork. I made a script that goes to each revision, runs the Clojure code, and runs OpenSCAD on certain of the resulting scad files. The next step was to render those into pictures, but I haven't gotten there yet. Some generalization of this is probably how to catalog the variants, rather than attempting to harmonize codebases made by people whose incentives are toward divergence not convergence.

jgwinner, 2018-10-26:
Thank you!

I had seen that one, I liked the 'drop to the floor' housing too, and the thread inserts.

I guess as I'm pretty new to DIY keyboards, I'm also sort of asking "Should I" :)  I haven't seen any of these with a traditional return key, or ]'s, etc.

Anyway, thanks for the response ... it's this or a Kinesis Pro :)

I'll close the issue (it's more of a discussion than an issue). I'll play around with the CAD files.

Love parameterized code; I was a user of POVRay for exactly that reason a long time ago.

== John ==

jotomo, 2018-10-26:
Hi John, I'm also starting my journey into DIY keyboards. I'm currently using a TEK (https://www.trulyergonomic.com/store/truly-ergonomic-mechanical-keyboard-soft-tactile-kailh-cherry-mx-compatible-brown-keyswitches-229-english which has enter and backspace in the middle) at my desk and a Thinkpad on the go (and an MS Sculpt ergo keyboard at work, for the moment).
Interestingly, my hands / muscle memory seem to recognize what's under them and just work (after 2-3 of initial learning where I hit the wrong side). Pressing enter/backspace with thumb/index on one keyboard or using the pinky on the other just works like touch typing any old character key. This also works with symbols that moved from left hand to right hand (=+-_), with only a few errors. So that seems to be a simple matter of learning, at least for me. 
Therefore I'm optimistic this will extend to the layers concept, which I find intriguing: rather than moving your hands over to other keys (or stretching them to get there), move the keys under your hand (or rather their assigned symbol) by pressing a modifier. E.g. when completing a method name, rather than to have to move the right hand down over the cursor keys, your left hand presses a layer modifier to move the cursor keys under your right hand, which can stay in place.
I'm also using vim, which has the navigation layer, which also works, making me more confident using and adjusting to a dactyl keyboard will work for, while still being able to use a normal keyboard on a notebook. HTH.

jgwinner, 2018-10-26:
Hi Jotomo!

I hear you on the layers, but I frequently use CAD tools that tie one hand (at least) into the mouse to do sensitive movements, then you want to hit an FKey. That's why the top row of F keys (in row layout) are pretty attractive to me. I get you on mainly using Linux.

This was the keyboard I have: The freestyle 2: https://www.kinesis-ergo.com/shop/freestyle2-for-pc-us/ 

I wore the one at work AND the one at home out - the little nibs on he F and J keys are shot, and I frequetnly ytor ygetibh ygubh, (I think that was 'the wrong thing' shifted a letter to the center, LOL).

So, I figured if I'll replace it, I'm either upgrading to: https://www.kinesis-ergo.com/shop/freestyle-pro/ but thought I'd check into the Ergo DIY keyboards. Sounds like fun projects.