adereth/dactyl-keyboard #66: Loading dactyl.clj

lungsi, 2018-11-19:
I have clojure 1.8.0 and OpenSCAD 2015.03.3 installed. My `project.clj` is
```
(defproject dactyl-keyboard "0.1.0-SNAPSHOT"
  :description "A parametrized, split-hand, concave, columnar, erogonomic keyboard"
  :url "http://example.com/FIXME"
  :main dactyl-keyboard.dactyl
  :license {:name "Eclipse Public License"
            :url "http://www.eclipse.org/legal/epl-v10.html"}
  :dependencies [[org.clojure/clojure "1.8.0"]
                 [unicode-math "0.2.0"]
                 [scad-clj "0.5.3"]])

```
But when I run (after `lein repl`)
`(load-file "src/dactyl_keyboard/dactyl.clj")`

I get
```
WARNING: import already refers to: #'clojure.core/import in namespace: dactyl-keyboard.dactyl, being replaced by: #'scad-clj.model/import
WARNING: use already refers to: #'clojure.core/use in namespace: dactyl-keyboard.dactyl, being replaced by: #'scad-clj.model/use

CompilerException java.lang.RuntimeException: Unable to resolve symbol: fixed-tenting in this context, compiling:(/data/dactyl-manuform/src/dactyl_keyboard/dactyl.clj:181:33) 
```
Any help will be much appreciated.

dedalo, 2020-06-08:
same error here any news about this?

jaredjennings, 2020-06-08:
This means some code (at line 181 column 33) tried to use a symbol `fixed-tenting` that wasn't defined. Look for a line that starts with a semicolon and contains "def fixed-tenting," and remove the semicolon.

dedalo, 2020-06-08:
> This means some code (at line 181 column 33) tried to use a symbol `fixed-tenting` that wasn't defined. Look for a line that starts with a semicolon and contains "def fixed-tenting," and remove the semicolon.

WORKED! thanks, ended up uncommenting 4 definition lines.

Kosaro, 2021-03-12:
Why are these commented in the first place? Shouldn't the clj compile without needing an edit?